Carrot-collective/ERCF_v2 #10: Request to add XOL cutter+sensor

fryc88, 2023-12-23:
Hi,
I know it's only been released, but are there any plans on adding cutter and/or filament sensor to XOLs current build (current build isn't XOL2 by the way):
https://github.com/Armchair-Engineering/Xol-Toolhead/tree/main
My combo is Orbiter2 and Rapido2 uhf, I've looked into some files but my CAD skills are too poor still to get this done on my own :(

fryc88, 2023-12-31:
Anyone? Is it just me who's got that set?

Tyler8299, 2023-12-31:
Ill get with the team to try and work on something. I myself am no good at CAD. I just reviewed the files and noticed the XOL does not support  the X1C hot end.  It's going to take me some time to acquire all this stuff as I run the X1C in all my gear.

fryc88, 2023-12-31:
I would really appreciate it :) no rush with it to, would be just nice to see it at some point

fryc88, 2023-12-31:
> Ill get with the team to try and work on something. I myself am no good at CAD. I just reviewed the files and noticed the XOL does not support the X1C hot end. It's going to take me some time to acquire all this stuff as I run the X1C in all my gear.
Actually there's one for current xol build, however not official, they just shared it on xol discord
https://www.printables.com/model/622854-bambulab-hotend-mount-for-xol_toolhead

Bb06477, 2025-03-07:
https://www.printables.com/model/1214979-g2sa-sensor-adaptation-for-xol2-toolhead