Carrot-collective/ERCF_v2 #13: Gneu42 Block End mod issue

Benoitone, 2023-12-23:
@gneu42 I think in your `Filament_Block_End.stl` from your Reinforced Block End Mod the two little conicals holes are missing?
[image]

gneu42, 2023-12-23:
You are right, they are missing, but I did not remove them.
They were already missing on the part that I used as base for the modification.
They must have been added later on to the V2 final block end.
They are not needed on my setup, since the foot is not under them.
[image]

Benoitone, 2023-12-23:
> You are right, they are missing, but I did not remove them. They were already missing on the part that I used as base for the modification. They must have been added later on to the V2 final block end. They are not needed on my setup, since the foot is not under them. ![Capture](https://private-user-images.githubusercontent.com/59416685/292635935-21e256e9-7683-48c3-893c-ce83ea8d2da9.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzNDk5OTQsIm5iZiI6MTcwMzM0OTY5NCwicGF0aCI6Ii81OTQxNjY4NS8yOTI2MzU5MzUtMjFlMjU2ZTktNzY4My00OGMzLTg5M2MtY2U4M2VhOGQyZGE5LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIzVDE2NDEzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJjZTE5NmJiM2RiMGI0YjgzZmJlNmJkMmM3ZDljOTY0MTUzZjM3M2I2YWRkZjM2NTU4M2Q4N2QzMjEyNTg5NzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.axHqlZKgFwoWFoVwoZ8vRaW6OS3a7m8_BheyQWJIDq0)

Ok. it's the foot from the original SturdyBunny :)

gneu42, 2023-12-24:
I added the 2 missing conicals holes, so it can be use with the official V2 foot.

closed 2024-01-04 by MissileLaunchSystem