Carrot-collective/ERCF_v2 #19: [a]_Support_Arm_xN.stl rabbit placed 1 layer above the bed

mszulen, 2023-12-25:
Is this intentional? There's a single layer space between beed and the rabbit figure in the Support Arm file.
[image]

o1hitman1o, 2023-12-26:
@mszulen How are you getting bridging on the first layer. I can't see to get it only on the second layer and when I try to print it wants to fail.

mszulen, 2023-12-26:
It does print but it's not looking good as you can see.
[image]
[image]

o1hitman1o, 2023-12-26:
It does appear that maybe this was a mistake, other photos on discord show it being solid on the first layer...
[image]

mszulen, 2023-12-26:
then the file in STL folder is wrong/broken.

o1hitman1o, 2023-12-26:
Per Terrance on Discord it is supposed to be one layer higher to give a texture to the rabbit.

mszulen, 2023-12-26:
oh.. ok. Anyway... it won't be visible once assembled :) Not a big issue :D

o1hitman1o, 2023-12-26:
I guess the issue can be closed then.

closed 2023-12-26 by mszulen