Carrot-collective/ERCF_v2 #21: 2020 mounting parts

Anollya, 2023-12-26:
Please make cuts in the 2020 mounting parts for the T-Nuts. 
The parts dont hold in the 2020 extrusion.

wrong
[image]

ok:
[image]

affected parts:
Stls/Filament_Blocks/End_Bypass.stl
Stls/Gear_Box/Gear_Box_Front.stl
Stls/Filament_Blocks/[a]_End_Bypass_Foot.stl

Tyler8299, 2023-12-26:
You are using T slot 2020 and not V slot 2020 correct? Are you using Drop in nuts/Hammer T nuts or spin in nuts? Hammer T nuts are almost form fitting giving no locking force please try drop in nuts

Tyler8299, 2023-12-26:
[image]

Tyler8299, 2023-12-26:
Here is an example of how both nuts engage the extrusion and if using V slot will compound the problem even worse [image]

Tyler8299, 2023-12-26:
[image]

ZeraLes, 2023-12-26:
Thank you Tyler, I. Using only the the ones for drop in with a spring lick and steel ball. I'm coming from the mechanic sides. And yes only t nut not vslot. Having my problems collecting 4050 double t nut here in Germany for my EZBake trident Projekt for 24/25 😬. Thanks a lot for the info, already ordered a 2020 on my fav shop here. Regards Boris --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

Anollya, 2023-12-26:
Hello Tyler,
thank you for your fast response.
I´m using V-Slots. I tested drop in nuts as well, the same.

i measured it:

[image]

[image]

[image]

V-Slot ~1,4mm - 0,7mm (drop in nut) = 0,7mm -> the part has 1,1mm -> 0,4mm to much.
i dont know, may be i bought the 2020 from a wrong dealer.

tinyfluffs, 2023-12-26:
I do personally second this change. Having the option to use hammer heads or roll in nuts would be greatly appreciated.

arkeet, 2023-12-30:
I noticed this too when trying to assemble with cheap slide-in nuts with a 1.2mm step (but with Misumi T-slot). I think 1.1mm on the part is excessive - for reference, I checked some similar features on a few Trident parts, and they have just 0.6-0.7mm extending into the slot, so I think that is sufficient.

The filament blocks have a cutout so they don't have this issue.

WizardUli, 2024-04-04:
I have this profile https://www.vsk-profily.cz/hlinikovy-profil-bh-20-6-20x20/ with spare nuts from LDO Voron kit and it's completely unusable. Gearbox and filament blocks are still completely loose after tightening them.