Carrot-collective/ERCF_v2 #23: CottonTail buffer - slicing issue.

bartoszwas, 2023-12-26:
Hey all,

In CottonTail buffer is hard to print that rabbit in the middle 'cause it starts on second layer... 
[image]

o1hitman1o, 2023-12-26:
According to Terrance on discord it is meant to be that way to add texture. He suggested you can also just [add this rabbit](https://cdn.discordapp.com/attachments/1165841694273585234/1189292318038184016/a_Support_Arm_Mount_Bunny_xN.stl?ex=659da1c8&is=658b2cc8&hm=2c2cd874c1406ed633cfc74089253dd92d6cdf4890480deec95a9198ede11d99&) in 
with this part, and make it flush to the plate. I just edited the STEP file to pull it to the surface so mine prints correctly. Attached are the files is used. 
[Support_Arm.zip](https://github.com/Enraged-Rabbit-Community/ERCF_v2/files/13773998/Support_Arm.zip)

o1hitman1o, 2023-12-26:
I feel this should be made as a 2 part object instead of a reces. That way if you wanted to print this as multicolor you could.

bartoszwas, 2023-12-26:
Yeah, multicolor would be a good option. Better then printing that complicated model in mid air...

w35t13, 2023-12-28:
if it was designed abit better maybe it could work but how it is just makes the hole rabbit print midair.

o1hitman1o, 2023-12-30:
@bartoszwas Looks like and updated support arm was added so it easier to print. If this works I think this issue could be closed.

closed 2024-01-02 by bartoszwas