Carrot-collective/ERCF_v2 #24: Documentation issue on page about Wiring the MMB

jaketri, 2023-12-27:
The manual page about Wiring the MMB (page 125 on current version of the documentation PDF) have pre-gate sensors using MMB connectors STP1 - STP8 and also shows jumpers ON for DIAG1 and DIAG2 (gear and selector motors). Based on [MMB documentation.](https://github.com/bigtreetech/MMB/tree/master/Hardware) STP1 / STP2 and DIAG1 / DIAG2 share same GPIOs (PA3 / PA4). Also the STP9 and STP10 are not used so a quick fix is to shift the pre-gate sensors to use STP3 - STP10 and reserve PA3 and PA4 GPIOs (used by STP1 and STP2) for DIAG1/2 signals.

[image]

o1hitman1o, 2023-12-29:
This seems correct but also their might be something wrong with MBB documentation. I see that STP3(STOP-3) connects to pin PB8 but  STP4(STOP-4) connects to pin (PB7) but on the Schematic is shows from the MCU section PB9 is STOP-3 and PB8 is STOP-4, PB7 is M4-DIR. 
Upon further research It seems this is an issue in the documentation on MMB [Link](https://github.com/bigtreetech/MMB/issues/7)
Overall I think you are correct with this (if you plan on using sensorless homing for any of the motors), you will need to leave STP1 & STP2 open. If not then you would not use the jumper pin on DIAG1 or DIAG2 and can use STP1&2 for the pre-gate sensors.

o1hitman1o, 2024-01-02:
@jaketri  looks like this was just updated. Close issue?

jaketri, 2024-01-02:
Closing issue as fixed by latest documentation update

closed 2024-01-02 by jaketri