Carrot-collective/ERCF_v2 #29: [Documentation] Missing instructions for Easy BRD install, and other boards

tinyfluffs, 2023-12-31:
Had to figure this out for myself. There's no CAD for Easy BRD either, so it wasn't immediately obvious where and how to mount the board. Would be good if there was a document on the Git repo at the very least, but ideally incorporated into the manual.

spikeygg, 2023-12-31:
This is a duplicate of https://github.com/Enraged-Rabbit-Community/ERCF_v2/issues/16

tinyfluffs, 2024-01-01:
> This is a duplicate of #16

That's my own. And no, it's not a duplicate.

sharpner, 2024-01-30:
@tinyfluffs could you share how you did it? I am currently stumbling over this as well^^"

tinyfluffs, 2024-01-30:
> @tinyfluffs could you share how you did it? I am currently stumbling over this as well^^"

Sure! Here's some photos from my build log. Basically, two t-nuts for the mounting frame. Then it's M3 screws threaded into plastic for the MCU.

[image]
[image]
[image]

sharpner, 2024-01-31:
awesome thanks so much - looks super clean :)

tinyfluffs, 2024-02-01:
> awesome thanks so much - looks super clean :)

No problem! One thing I do recommend is running a separate 5V rail to your servo. The Easy Brd 5v rail is not powerful enough to handle ERCFv2, as it needs extra current. If you have a 5V PSU you can just run the single cable from there, provided the grounds are tied together with the 24V.