Carrot-collective/ERCF_v2 #30: BOM wrong for 5.5 Ball

Mac2712, 2023-12-31:
BOM shows 5.5mm Ball for CW2 main body for filametrix looks like a 7mm ball is needed for the bearing switch.

o1hitman1o, 2024-01-02:
I agree with this. BOM says 5.5 but Filiametrix says 7mm.

bartoszwas, 2024-01-02:
Yeah that filamterix is a mess, every variant is different and propably cable cover won't fit either...

Mac2712, 2024-01-02:
I purchased some ball's just for this because it was on the BOM in the end I printed a ball shaped spacer that works nice

kinematicdigit, 2024-01-04:
We are investigating this further, and yes, some of the mods have 7mm balls. We are trying to make them all standardized to 5.5 balls. Apologies for that. We are going to address this as soon as possible and will post updates. In the meantime, if you can find 7mm balls, use them, but we will be moving forward with one standard.

kinematicdigit, 2024-01-04:
> Yeah that filamterix is a mess, every variant is different and propably cable cover won't fit either...

We did try to ensure the covers and latches all match. We will double check those as well.

EmJay276, 2024-01-17:
@kinematicdigit let me know if you need any changes in these parts here
https://github.com/Enraged-Rabbit-Community/ERCF_v2/pull/7