Carrot-collective/ERCF_v2 #32: New easier to print [a]_BTT_MMB_Support_Arm_Endcap_Bunny_Outline.stl does not have cutouts for alignment with coupler blocks.

Ekodas, 2023-12-31:
The new easier to print [a]_BTT_MMB_Support_Arm_Endcap_Bunny_Outline.stl does not have the 3 alignment cutouts to allow it to mate with the coupler blocks.  The old version did have these cutouts.

o1hitman1o, 2024-01-02:
- [ ] 
[image]
[image]
This is confirmed. Need to update easy to print version.

MissileLaunchSystem, 2024-01-04:
Resolved with a commit I just pushed a minute ago. The new file is available here:
Recommended_Options/ERCT_Buffer/Stls/Support_Arm/Bunny Logo Outline (easier to print)/[a]_BTT_MMB_Support_Arm_Endcap_Bunny_Outline.stl

closed 2024-01-04 by MissileLaunchSystem

theminjus, 2024-01-06:
i would reopen this one , 
[a]_Support_Arm_Endcap_No_Electronic_Box_Bunny_Outline.stl
is hole is not aligned for the endcap heat insert