Carrot-collective/ERCF_v2 #45: Apron LEDs

Zephyrus67, 2024-01-14:
Does the apron leds do anything besides throwing light inside the Voron printer through the top panel?

moggieuk, 2024-01-19:
Yes!  There are certain effects that occur for operations but the functional part is that they can show the filament color or gate_state .. gate have filament or not as well as indicate error conditions and filament swaps.  It certainly isn't necessary but a nice feature and more than just lights.  Here is an example through the HH command:
```
> MMU_LED
LEDs are enabled
Default exit effect: 'gate_status'
Default entry effect: Unavailable
Default status effect: 'filament_color'

Options:
ENABLE=[0|1]
EXIT_EFFECT=[off|gate_status|filament_color]
ENTRY_EFFECT=[off|gate_status|filament_color]
STATUS_EFFECT=[off|on|filament_color]
```

Zephyrus67, 2024-01-19:
Thank you very much. I already have case lighting so I would not use the aprons if all they did was throwing light into the case. But in this case as you describe I will print out the aprons as well.

I am not 100 % certain yet where I will place the ERCF (might go on the back of the printer) so I might make a new standalone panel for the aprons instead but I will see that later.

closed 2024-01-21 by Zephyrus67