Carrot-collective/ERCF_v2 #61: Integrated support is not printed correctly

mboo83, 2024-02-07:
The support is only partially printed. Different slicers and settings show the same result. 

[image]
[image]

alexstone, 2024-02-08:
I printed with Arachne wall generator and managed to get the support to print. However, without this setting, those supports straight up _do not_ print and you end up with blocks you can't really use.

I would prefer to see the block without the in-build supports, and markers in the docs of where to add them. This would allow the use of interface materials for cleaner bottom surfaces.

mboo83, 2024-02-09:
Thank you, i will give it a Try. You can export it from the CAD Version with Fusion 360 and remove the Support. I printed the one that comes with the Thumper Blocks Mod. That did came out perfect with the integrated Support.

MissileLaunchSystem, 2024-03-13:
I've added support-less versions of the Filament_Path, Tophat, and Base under the sub-folder "No_Supports."

closed 2024-03-13 by MissileLaunchSystem