Carrot-collective/ERCF_v2 #63: request for a file to put a list of parts

StalkFPV, 2024-02-08:
english:

and they should have a read me with the list of materials needed for the Project and also the number of pieces to print example for 4.

because I am French and frankly I have the complete file and I am totally lost because I don't even know which part exactly goes where and the number of parts to print.

so it would be great if the creator or someone from the community helps people who are lost or new

anyway great job

french :

et ils faudrait avoir un lisez moi avec la liste de matériel qui faut pour le Project et aussi le nombre de pièce a imprimer exemple pour 4 .

car moi je suis français et franchement j'ai dl le dossier complet et je suis totalement perdu car je c'est même pas quel pièce exactement qui va ou et le nombre de pièce a imprimer .

du coup cela serrait le top que le créateur ou une personne de la communauté aide les personne qui sont perdu ou nouvelle

en tout cas super boulots

olivbd, 2024-02-08:
The "list of materials" is called "BOM" - Bill Of Material. You can find it on the [README.md](https://github.com/Enraged-Rabbit-Community/ERCF_v2/blob/master/README.md) file of this project

[image]

StalkFPV, 2024-02-08:
thank you, I hadn't even seen it. then we cannot modify the number of blocks, but I made a request.

olivbd, 2024-02-08:
> thank you, I hadn't even seen it. then we cannot modify the number of blocks, but I made a request.

When you open this spreasheet, it's in read-only mode. You have to make a copy (File -> Make a copy) and then you will able to modify the number of filament blocks.

[image]

StalkFPV, 2024-02-08:
thank you damn really google drive is really not my strong point

olivbd, 2024-02-08:
You're welcome.

Could you close this issue?