Carrot-collective/ERCF_v2 #64: Suggest adding more clearance to the gear box to route motor wires through

alexstone, 2024-02-08:
I wanted to replace my NEMA14 with a NEMA17. When trying to remove the NEMA14, I found that I actually had to disassemble the _entire_ ERCF to get the cable out without snipping it and re-doing the housing. An extra few millimeters here would really help with maintenance.

[image]

mboo83, 2024-02-09:
I did the same Mistake. You can easyly pinout the Cables and pull them through the Hole.

MissileLaunchSystem, 2024-03-13:
I've merged Ark's changes that add more space to the wiring path. This is as fixed as it is going to get until we redesign the gearbox.

closed 2024-03-13 by MissileLaunchSystem