Carrot-collective/ERCF_v2 #67: ERCT Overhanging Rabbit

3DPrintDemon, 2024-02-14:
The rabbit logo on file [a]_BTT_MMB_Support_Arm_Endcap.stl is an overhanging surface of 0.25mm off of the build plate, Orca/Prusa slicers will not support it.
[image]
[image]
[image]
[image]

o1hitman1o, 2024-02-14:
@3DPrintDemon it's is definitely and issue and it would work if the slicer realized it was unsupported and bridge printed it but the slicers I have used wants to print the outline of the shape first and the edges at the curves want to print in mid air. 
This is the solution though.
https://github.com/Enraged-Rabbit-Community/ERCF_v2/tree/master/Recommended_Options/ERCT_Buffer/Stls/Support_Arm/Bunny%20Logo%20Outline%20(easier%20to%20print)

3DPrintDemon, 2024-02-14:
> @3DPrintDemon it's is definitely an issue and it would work if the slicer realized it was unsupported and bridge printed it but the slicers I have used wants to print the outline of the shape first and the edges at the curves want to print in mid air.
> This is the solution though.
> https://github.com/Enraged-Rabbit-Community/ERCF_v2/tree/master/Recommended_Options/ERCT_Buffer/Stls/Support_Arm/Bunny%20Logo%20Outline%20(easier%20to%20print)

Hi. Yes it needs an outline on layer 1 then it would be able print the rabbit as an unsupported bridge, however as you say the outline starts on layer 2 with nothing under it so it prints in mid-air. 

Indeed, I mentioned about those outline files, it would be good to include them with the files at the link I posted above along with the main ERCF files, as it is they’re only available deep in the folders of the ERCT files, not that obvious to people unfamiliar with the directory structure & find only the regular ones in the ERCF folder first. 

Tbh it would probably be best to take the regular files down or fix them as Orca & Prusa simply won’t print them correctly.

wagnerism, 2025-03-09:
This is causing failed prints in both PrusaSlicer and OrcaSlicer.  The first one or two worked. I had several failures until I looked at the second layer.  There should be an outline of the rabbit on the very first layer.

This is a poorly-implemented and unnecessary cosmetic feature.

STL uploads are not allowed here so I filled in that first 0.25mm layer and uploaded it here.  https://github.com/wagnerism/tamed_rabbit/blob/main/%5Ba%5D_Support_Arm_tamed_rabbit_of_death_xN.stl

Here's the first layer:

[image]