Carrot-collective/ERCF_v2 #79: Filament_Path_xN.stl has 6 non-mainfold edges

relder, 2024-03-13:
Both prusaslicer and orca slicer complain that https://github.com/Enraged-Rabbit-Community/ERCF_v2/blob/master/Stls/Filament_Blocks/Filament_Path_xN.stl has 6 non-mainfold edges and recommends repair.

MissileLaunchSystem, 2024-03-13:
It's a known issue with the built-in supports. If you fix the manifold edges, it makes the supports print as a solid piece instead of hollow, which uses (read: wastes) about a gram more plastic. I'll ask the designer to take a look again.

gneu42, 2024-03-14:
Yes, this know since the very begining. I think it is a fusion 360 bug.
Also, make sure to have the detect thin wall ticked, or you may have no supports.

Just ignore it, as mentioned in the readme file from triple decky github: https://github.com/gneu42/Triple-Decky?tab=readme-ov-file#printing-tips

[image]

closed 2024-03-28 by gneu42