Carrot-collective/ERCF_v2 #91: FR - Galileo 2 Extruder with (2) filament sensors

77arkus76ohr, 2024-04-09:
Hi all,
I was wondering if someone worked or is working on a MOD to have two filament sensors for the new Galileo 2 extruder (G2E) by @jaredC01? He pointed out in his Issue-tracker, that ERCF_v2 is dealing with it?!? https://github.com/JaredC01/Galileo2/issues/36#issuecomment-2030825242

I did not find a hint for the G2E in here. Did I miss it?

I was checking with the last STL of the G2 how to fit a D5H to it. But I wasn't successful. Yesterday I came across the D2H from omron. Maybe this can make it possible? Or did someone else found a solution already? https://omronfs.omron.com/en_US/ecb/products/pdf/en-d2hw.pdf

igiannakas, 2024-04-09:
There are two ways to do this:
1. With d2f switches and bearings: https://github.com/IRTrail/G2E-Filametrix
2. with a d2f lever switch without bearings: https://github.com/rongith/Adaptation-from-Tircown-s-filament-sensor (it replaces the front block).

im using option 2 - less filament drag and an easier print.

both have been updated with galileo 2 final release.

Tyler8299, 2024-04-09:
Something like this?

Tyler8299, 2024-04-09:
https://github.com/IRTrail/G2E-Filametrix

igiannakas, 2024-04-09:
> https://github.com/IRTrail/G2E-Filametrix

Personally I’m a bigger fan of this front plate: https://github.com/juliusjj25/G2E-Filametrix-Lever-Switch-Mod

as it doesn’t use ball bearings there is less friction in the filament path and it’s less sensitive to micro switch tolerances.  Printed both but ended up with the above.

Tyler8299, 2024-04-09:
I’ve not experienced any issues or added friction in my 3 toolheads

https://github.com/IRTrail/G2E-Filametrix

Tyler8299, 2024-04-09:
sorry I was a bit asleep responding last night here is the version the ERCF team recommends using.  https://github.com/IRTrail/G2E-Filametrix Silverback Attack "ERCF team member" has been working hard making a Flawless Dual Sensor version which has been extensively tested by the team to not create any false positives. Enjoy!