Carrot-collective/ERCF_v2 #107: request to support larger and stronger springy servo

ntchris, 2024-06-20:
request to support larger springy push down servo

feature request to support 28mm length servo or even larger servo ?
thanks.
currently it only supports 23mm servo (22.5mm)
I printed and installed springy for a few months now,
I found that the 23mm wide MG90S micro servo sometimes cannot move to "down" position, because the spring resists and move the arm back.
Maybe the 23mm wide MG90S micro servo is too weak, and cannot provide necessary force to push down the strong spring.
The work around is to tweak with the spring length, cut and try.
and I also have to use 0.8mm dia spring.

There are some slightly larger 28mm wide servo in market with cheap price, should be stronger and better.
for example
https://www.aliexpress.com/item/1005003643468291.html?spm=a2g0o.cart.0.0.5cfb38dapkgguq&mp=1
17g metal gear

Maybe 33mm servo is even better,   
or,  36mm or 40mm servo is too large to fit ?
Thanks!

This request was originally opened for Springy project because no step file is available and seems that project was dropped/abandoned.
Now that we have ERCF V2 , the frame is much stronger, and the CAD files are available....seems should make more sense and easier.
Just open this request maybe many other users need it as well.
Thanks.

 
[image]

Thanks

MakManIfest, 2024-06-26:
You should be able to find a higher torque servo of the same dimensions as the MG90S.

Servo's mostly have semi standardized dimensions. For the MG90S dimensions you should look for high torque digital servos for use in 450 size R/C helicopter tail and swashplate.

These servos will be high speed and very precise with plenty of torque, 3-4kg. They will either be all metal or mostly metal construction for heat dissipation as the down side of these servos is they usually require high voltage to get the max torque and speed. By high voltage I mean 6-8 volts. But there are some that will run happily at 5v with decreased torque and speed.

During my time flying R/C heli's I tested many different servos and found you get what you pay for. You can save some money by getting a clone of a KST servo. Like this for example. https://www.aliexpress.com/item/1005005625391299.html

ntchris, 2024-08-15:
done.
https://www.printables.com/model/971423-ercf-v2-selector-mod-for-larger-28mm-17g-servo

https://github.com/ntchris/voron_mod_super_diy/tree/main/ercf_v2/servo_28mm_selector

[image]

[image]