Carrot-collective/ERCF_v2 #140: [Sugestion] Coupler Block with 4.5 mm steel ball

klone52, 2024-09-29:
There are cheaper and easier to find 4.5 mm steel balls in airsoft ammunition, with great diameter consistency. For this you would have to move the pre-gate switch closer and decrease the diameter of the hole by 1 mm.

To make it backward compatible with the kit boards you would have to bring the switch closer and make space in this way:

[image]

Tyler8299, 2024-09-29:
Have u had issues? I have (6) 8 gates built for testing and the bearing and switches have never been an issues

klone52, 2024-09-29:
i bougth my kit from siboor and came with 1 ball bearing 8 mm diam. I'm from Chile and is not easy to find 5.5 mm steel balls. The best alternative i found was the bbs from airsoft guns, and they are pretty cheap , like 500 for  5 USD

Tyler8299, 2024-09-29:
I'm sorry to hear that. Something so simple and readily available mostly around the world. That really does make things a bit difficult for you but most of us can get this overnight or even the same day CHEAP within good tolerances. Have you made the modification already?

klone52, 2024-09-29:
I made it for me and i'm waiting to have spare time to test it. I think in a couple of days i can have it installed on my 8 gates

Tyler8299, 2024-09-29:
Best of luck please keep us posted on how everything works out