Carrot-collective/ERCF_v2 #153: Filament and tube generate static electricity then cause mmu board been shock and disconnect. (Potential damage)

willzhai55, 2024-11-18:
I noticed that when mmu loads or unloads filament, the friction can cause an electricity shock to mmu and cause it to disconnect or even stop working. The shock will happen between the tube and the encoder. My encoder stopped working after a few shocks. My solution was to ground the bearing. After this modification, the mmu has never stopped working. I was shocked when I removed the tube form ercf and realized this was the issue. 

It might be because I am in Canada and the humidity is around 20%. I did two ercf with three different motherboards It will happen all the time until I ground the bearing. (Also on my voron 0.2 it will shock the filament sensor and cause the mcu to die. Even the soc in some cases. I was absolutely no ideal before and now everything is just much more stable.)

tomlikesrocks, 2024-11-21:
Are you feeling or seeing these static shocks?
I'm in Perth Australia where our humidity is also very low, but I've never experienced a shock before.
However, I have had some disconnection issues.

DarkShadowsX5, 2024-12-04:
You can ground it out by wrapping uninsulated wire around where ever the static is building up and connect it to a ground. 
or print it in ESD safe filament since you're in an area prone to Static build up.

SkiBikePrint, 2025-01-04:
I can confirm this is a legitimate issue as I experienced this multiple times back when I used the ERCF EZ board. As an electronics manufacturing engineer, managing static electricity buildup and discharge (ESD) is very important for protecting sensitive semiconductor devices.