Carrot-collective/ERCF_v2 #171: Filament tied in ERCT Buffer

kktonev, 2025-03-27:
I try different buffer wheels and aslo the experimental:
Recommended_Options\ERCT_Buffer\User_Mods\michaelw7177\Bufferwheel_Long_Bowden_Experimental

I use it with 5 winds on Voron 2.4.

When load it manually - no problem on the firs gate use. After first use of the gate and turn back the filament is going wind over another.
Same happen when it is a long print with same filament. Till now the modified version is better, but happen friquently to become too thight layers and the extrusion became non stable a little bit.

Next thing is to try to shorten the bowden a little bit and make it work with only 4 winds. Should be better.

If any other solution will be happy to try it.

norris108, 2025-03-27:
The rule of thumb for the number of loops around the buffer wheel is:

Bowden Length / 230 rounded up

How long is your bowden?

kktonev, 2025-03-27:
> The rule of thumb for the number of loops around the buffer wheel is:
> 
> Bowden Length / 230 rounded up to the next whole number
> 
> How long is your bowden?

Probably didn't measure it correct - 750-800mm.  But 4 winds are too short - after eject the filament has few centimeters outside the buffer. There is some reserve to cut a piece of the tube and to stay with 4 winds.
Filament has some tension, because it comes from dryer inside of PTFE tubes, but this tension is not so much to make wind overlaping.

[image]